iforce2d/weenyPRU #2: Trying to run on pre-build Raspberry pi LinuxCNC distro.

Jaron123, 2024-02-15:
Hello, I'm trying to run your your weeny project on a Raspberry pi 5 using the pre-made linuxCNC distro file from the LinuxCNC website. I'm not sure what is causing the problem but running the hallcompile and install of weeny.c goes without problems. I have a bluepill STM32F103C8T6 programmed with weenyPRU-default.bin and wired them as in the weenyPRU v1.2 image on oshwlab website (Connected to SPI0).

When I then try to run linuxcnc with weeny-xyza.ini and hal example files i've gotten different errors at different times but it never works. 

The first error I got was this:

```
linuxcnc weeny-xyza.ini
LINUXCNC - 2.9.1
Machine configuration directory is '/home/cnc/Downloads/weenyPRU-main/configsamples/weeny-xyza2'
Machine configuration file is 'weeny-xyza.ini'
Starting LinuxCNC...
emc/iotask/ioControl.cc 786: can't load tool table.
linuxcnc TPMOD=tpmod HOMEMOD=homemod EMCMOT=motmod
Note: Using POSIX realtime
Found file(REL): ./weeny-xyza.hal
Waiting for component 'weeny' to become ready..................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
```

Then linuxcnc won't shut down properly so I have to reboot, at some other try I got this:
```
linuxcnc weeny-xyza.ini
LINUXCNC - 2.9.1
Machine configuration directory is '/home/cnc/Downloads/weenyPRU-main/configsamples/weeny-xyza'
Machine configuration file is 'weeny-xyza.ini'
Starting LinuxCNC...
emc/iotask/ioControl.cc 786: can't load tool table.
linuxcnc TPMOD=tpmod HOMEMOD=homemod EMCMOT=motmod
Note: Using POSIX realtime
Found file(REL): ./weeny-xyza.hal
link (updating variable file): No such file or directory
note: MAXV     max: 27.000 units/sec 1620.000 units/min
note: LJOG     max: 27.000 units/sec 1620.000 units/min
note: LJOG default: 35.000 units/sec 2100.000 units/min
note: AJOG     max: 45.000 units/sec 2700.000 units/min
note: AJOG default: 36.000 units/sec 2160.000 units/min
note: jog_order='XYZA'
note: jog_invert=set()
X Error of failed request:  BadWindow (invalid Window parameter)
  Major opcode of failed request:  3 (X_GetWindowAttributes)
  Resource id in failed request:  0xffffffff
  Serial number of failed request:  660
  Current serial number in output stream:  661
Shutting down and cleaning up LinuxCNC...
task: 65 cycles, min=0.000009, max=0.010198, avg=0.007815, 0 latency excursions (> 10x expected cycle time of 0.010000s)
Note: Using POSIX realtime
LinuxCNC terminated with an error.  You can find more information in the log:
    /home/cnc/linuxcnc_debug.txt
and
    /home/cnc/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
```

and then at another try (Maybe this was after I rebooted after linuxcnc didn't shut down properly, not sure)
```
linuxcnc weeny-xyza.ini
LINUXCNC - 2.9.1
Machine configuration directory is '/home/cnc/Desktop/weenyPRU-main/configsamples/weeny-xyza'
Machine configuration file is 'weeny-xyza.ini'
Starting LinuxCNC...
emc/iotask/ioControl.cc 786: can't load tool table.
linuxcnc TPMOD=tpmod HOMEMOD=homemod EMCMOT=motmod
Note: Using POSIX realtime
Found file(REL): ./weeny-xyza.hal
./weeny-xyza.hal:11: Pin 'weeny.SPI-enable' does not exist
Shutting down and cleaning up LinuxCNC...
Note: Using POSIX realtime
LinuxCNC terminated with an error.  You can find more information in the log:
    /home/cnc/linuxcnc_debug.txt
and
    /home/cnc/linuxcnc_print.txt
as well as in the output of the shell command 'dmesg' and in the terminal
```

All I did was re-try after forcefully rebooting the pi after linuxcnc wouldn't shut down properly. i'm not sure the errors say anything usefull but hope you can help.

Oh and I wandered something, do I have to change anything to config.txt? I enabled SPI0 manually since it seemed only logical linux cnc would need it but it doesn't say anywhere in the documentation.

iforce2d, 2024-02-16:
My guess is the SPI driver used by the weeny component is not supported. The RPi5 have a quite different arrangement of peripherals to previous models. The code for the weeny component uses bcm2835 driver which is not suitable either. So if this is the problem, it would explain why the component compiles ok but has problems running. It looks like the Pi5 is unsupported for a while.
https://forum.linuxcnc.org/9-installing-linuxcnc/50203-linuxcnc-on-raspberry-pi-5?start=70#287736

If you have a 3B or 4B you would have much more success with those. fwiw all my setups are standard Buster, then replace the kernel with a realtime one, then build LinuxCNC from source. Those last two steps are quite slow and tedious to do in full, so I have some pre-compiled files I use. If you're only thinking to use the Pi for LinuxCNC then I think Buster would be good enough. I'll outline the steps here in case others might be interested.

I downloaded the Buster install from here, unfortunately it looks like they've scrubbed it now, wtf?
https://www.raspberrypi.com/software/operating-systems/
Anyway the filename is '2023-02-21-raspios-buster-armhf.img.xz', maybe you can find it somewhere (plz link here if you do).

Full instructions for compiling a realtime kernel from scratch can be found here:
https://lemariva.com/blog/2019/09/raspberry-pi-4b-preempt-rt-kernel-419y-performance-test
Or if you want to use my pre-built files you can skip to the part where the kernel is actually installed (search for 'tgz' on that page). You can find my files here, note that they are different for 3B/Zero and 4B !
https://iforce2d.net/weeny

To install LinuxCNC after that you can use my pre-built .deb files, also at the above link.

By the way, you should normally be able to stop the LinuxCNC subsystem without a reboot (although who knows what the unsupported driver on Pi5 was doing). I have a script here which I use to shut down when running headless (no GUI). 
https://github.com/iforce2d/linuxcnc-gcode-server/tree/main/headless
Failing that you could try killing the processes 'rtapi_app', 'milltask' and maybe also 'axis'.

Also btw, the weeny component should start up ok even without any STM32 hardware attached. It's only when you turn estop off that the SPI communication begins. When using a bluepill myself, I found that using only one ground connection between the Pi and the STM32 was sometimes not enough for a reliable connection, so I would recommend at least two grounds.

Jaron123, 2024-02-16:
Thank you very much for the quick reply! It looks like on the raspberry pi 5 the most direct way of using the spi is via spidev.h would it be theoretically possible to rewrite the weeny.c to use spidev instead of bcm2835?
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#spi-overview
https://github.com/torvalds/linux/blob/master/drivers/spi/spi-bcm2835.c
https://raspberrypi.stackexchange.com/questions/123741/how-is-spidev-and-the-spi-bcm2835-module-related

It looks like weeny uses rt_bcm2835_init instead of normal bcm2835_init by using rtapi_open_as_root instead of open, i'm not sure if spidev can do something like that or if it can possibly do without?

iforce2d, 2024-02-17:
To be honest I don't know much about the SPI setup, or spidev. That part of the code I just used unchanged from the original Remora project. You might be able to find more knowledgeable answers in this forum thread, although I don't know if much is being done with Pi5, and the main developer seems to be more focused on ethernet methods these days:
https://forum.linuxcnc.org/18-computer/42276-remora-rpi-software-stepping-using-external-microcontroller-via-spi

iforce2d, 2024-04-19:
Just dropping this link here for myself in future, looks useful:
https://github.com/praktronics/rpi5-rp1-spi

iforce2d, 2025-07-24:
Finally got around to this. You should be able to use the HAL component on RPi5 now. I used the image from here: https://forum.linuxcnc.org/9-installing-linuxcnc/55192-linuxcnc-the-raspberry-pi-4-5-official-images-only

(specifically: https://www.linuxcnc.org/iso/rpi-5-debian-bookworm-6.12.11-arm64-ext4-2025-01-27-0140.img.xz )

closed 2025-07-24 by iforce2d