iforce2d/weenyPRU #5: Question: SetUp LinuxCNC card

guido-30, 2024-11-19:
Sorry if I write in the issue, Congratutation on the project. I could try use the your project board PRU for my project lathe DIY.
I use on RPI4 linux 6.1.54-rt15 PREEMPT_RT of 17-november 2023.
Next week I build the board whit STM32F1xx.
My questions are:
In OS Linux I must intall other driver for SPI communication?
Which board I can select in LinuxCNC
Many thanks in advance
Guido

iforce2d, 2024-11-19:
I recently installed one of these in my router, I made a basic explanation of the steps here: https://youtu.be/HaIe76u16mM?t=406
See also the 'setupStepsPi4.txt' file in the main folder.

I have never needed to enable SPI with any of the installs I've tried, but I can't speak for that specific one you're using.

I don't think any of the options in the LinuxCNC setup wizard will be useful. You'll need to start with the example in the 'configsamples' folder and change it as necessary for a lathe.

guido-30, 2024-11-24:
many thanks,
I saw the video, very interesting. Great.
In your opinion, is it possible to do a thread on the lathe with tool with open loop sepper motors without encoder with Linuxcnc and your STM32 board?

iforce2d, 2024-11-25:
I've never done that manually let alone by CNC, but I'm pretty sure the timing between the spindle and the cutter needs to be synchronized properly. It would only work with some kind of encoder, or maybe if the spindle was also being moved by a stepper motor.