iforce2d/weenyPRU #6: Out STEP3 no clean

guido-30, 2025-01-09:
Hi,
sometime in the stop move, after move axis, on the CPU pin 12 STEP3 (in my case axis Z) is present the signal of 5 microsecond every  about 5mSec.
The other asix are without this problem.

The move axis linuxcnc are in stop mode with velocity null. The signal on CPU pin is not regular it seems dirty but the pulse of 5 uSec seems corret and generate of CPU timer

Is already success?
Many thanks
Guido

iforce2d, 2025-01-09:
What is the DIR pin doing during this time? I have seen the output value jittering around the setpoint, where the stepper is being instructed to continually move back and forward one step. In my case this had no adverse effect on my machine (one step was not enough to even see any movement). I think it might be something to do with floating point precision and how the DDS accumulator works, but I don't quite understand it fully.

The Remora code was altered to use 'rawCounts' which seems to prevent this for the most part. I updated weenyPRU firmware around November 16-18 with the same change. If you're using an older firmware, perhaps you could try the latest version. Please note that you'll also need to update the LinuxCNC component as well.

guido-30, 2025-01-09:
thanks for your asnware,
This issue I saw only on the axis number 3 on the axis 1 and 2 are done.
In my case the motor rotate in low speed and not constant speed, as if it were a disturbing noise 
I used the your build 1.4, now check the linux component and retry.
Many thanks

iforce2d, 2025-02-12:
How did that go?