iforce2d/weenyPRU #8: Use a encoder & selector axis JOG controller

guido-30, 2025-02-11:
Hi and congratulation for the project,
I would like connect a external box with encoder JOG speed controller and emergency stop at weeny board. 2 question:
1) is it correct to connect the encoder box to the board or is it better to connect it directly to the RPI?
I don't use the card drivers and the serial so my idea is to use the serial (or wireless) to communicate with the encoder box. So far so good. Does the RPI read the new positions of the axes?
2) In the case to connet the encoder box directly to RPI, you know if and how the linuxcnc can undestand the data recived ?
many thanks
guido

guido-30, 2025-02-14:
ok, 
I would try the encoder slow only for jog move the axes conneted EC11 (CLK-DT) encoder with 30 PPR on the PC14 & 15 pin with resistor o pull-up. If I understand how run, with downloaded the 1.4 release fw in micro , I made zeros at all axes. 
if I rotate the encoder I expect a increment or decrement the position value on axis select in the linuxcnc .

If I try to build the project in the repository the encoder no run same. 
do I have to do something special in the file source code?
thanks

iforce2d, 2025-02-14:
As I said above, rotating the dial will cause the weeny.encoder.jogcounts value in LinuxCNC to increment or decrement by 1. That is not the same as changing the axis position, I mean which axis should it control? You would need to configure your .hal file to enable jogging, link the jogcounts value, and set the movement scale. You can see a commented out example in the .hal file. Try un-commenting lines 80-83 here: https://github.com/iforce2d/weenyPRU/blob/main/configsamples/weeny-xyza/weeny-xyza.hal#L79

btw I don't think you need pull-ups for this, is there some reason to add them? Not that it should matter either way... I think.

Also, can you please take a look here and answer: https://github.com/iforce2d/weenyPRU/issues/6

guido-30, 2025-02-14:
perfect, many thanks.
I managed to buid the project without the UART before in debug and now in release. 
I anable the raw 80-83 and now th axis Z count..
A question: I can select in hal file the axis X or Y or Z with a input pin on the board? I like make a IF istruction with the PIN select in the hal file.

iforce2d, 2025-02-15:
I don't know how you could expect to select 3 options with just one pin....?

But you could select between *two* axes with one digital input. For example, axis.x.jog-enable would be set to true when the digital input is high, and axis.y.jog-enable would be set to true when the digital input is low (or NOT high, using the NOT component). 

This is a question about LinuxCNC in general, not weenyPRU. Maybe read this page to gain a good foundation in this stuff: https://www.linuxcnc.org/docs/2.5/html/hal/basic_hal.html

Also, can you please take a look here and answer: https://github.com/iforce2d/weenyPRU/issues/6